jontek2/A-better-print_start-macro #5: printer status and led effects?

RedSinkDirty, 2023-07-23:
Do you have to install led_effects to use the printer status macros? I don't know I already had it led effects installed. just realizing there isnt mention of that. maybe there doesnt need to be? But anyway I think you should change the name of this project to "The Best Print Start Macro" because it is, this was such a huge help, Im glad I found it, Thank you for sharing it.

jontek2, 2023-07-23:
Hey,

That's awesome to hear! 

No, you don't need to install any plugins. The different commands in the macro such as STATUS_HOMING is referring to the Stealthburner LED macros, se the macros here: https://github.com/VoronDesign/Voron-Stealthburner/blob/main/Firmware/stealthburner_leds.cfg

RedSinkDirty, 2023-07-24:
oh I didnt know about that, ive just always used https://github.com/julianschill/klipper-led_effect   which is like almost the same thing but with animations. I guess this must have came from that because the config is super similar and the macros are all named the same, so I thought the macros were for led effects.

closed 2023-07-24 by RedSinkDirty