jontek2/A-better-print_start-macro #7: heat soak under 90 bed temp

larkinnn, 2023-08-18:
how can i only get the printer to heat coak when its only asking for a bed tempter above 90c

this is how i have your macro sry up for myself
[gcode_macro PRINT_START]
gcode:
  # This part fetches data from your slicer. Such as bed temp, extruder temp, chamber temp and size of your printer.
  {% set target_bed = params.BED|int %}
  {% set target_extruder = params.EXTRUDER|int %}
  {% set target_chamber = params.CHAMBER|default("40")|int %}
  {% set x_wait = printer.toolhead.axis_maximum.x|float / 2 %}
  {% set y_wait = printer.toolhead.axis_maximum.y|float / 2 %}

  # Homes the printer, sets absolute positioning and updates the Stealthburner leds.
  STATUS_HOMING         # Sets SB-leds to homing-mode
  G28                   # Full home (XYZ)
  G90                   # Absolut position

  ##  Uncomment for bed mesh (1 of 2)
  BED_MESH_CLEAR       # Clears old saved bed mesh (if any)

  # Checks if the bed temp is higher than 90c - if so then trigger a heatsoak.
  {% if params.BED|int > 90 %}
    SET_DISPLAY_TEXT MSG="Bed: {target_bed}c"           # Displays info
    STATUS_HEATING                                      # Sets SB-leds to heating-mode
    M106 S255                                           # Turns on the PT-fan

    ##  Uncomment if you have a Nevermore.
    SET_PIN PIN=nevermore VALUE=1                      # Turns on the nevermore

    G1 X{x_wait} Y{y_wait} Z15 F9000                    # Goes to center of the bed
    M190 S{target_bed}                                  # Sets the target temp for the bed
    SET_DISPLAY_TEXT MSG="Heatsoak: {target_chamber}c"  # Displays info
    TEMPERATURE_WAIT SENSOR="temperature_sensor chamber" MINIMUM={target_chamber}   # Waits for chamber to reach desired temp

  # If the bed temp is not over 90c, then it skips the heatsoak and just heats up to set temp with a 5min soak
  {% else %}
    SET_DISPLAY_TEXT MSG="Bed: {target_bed}c"           # Displays info
    STATUS_HEATING                                      # Sets SB-leds to heating-mode
    G1 X{x_wait} Y{y_wait} Z15 F9000                    # Goes to center of the bed
    M190 S{target_bed}                                  # Sets the target temp for the bed
    SET_DISPLAY_TEXT MSG="Soak for 5min"                # Displays info
    G4 P300000                                          # Waits 5 min for the bedtemp to stabilize
  {% endif %}

  # Heating nozzle to 150 degrees. This helps with getting a correct Z-home
  SET_DISPLAY_TEXT MSG="Hotend: 150c"          # Displays info
  M109 S150                                    # Heats the nozzle to 150c

  ##  Uncomment for V2 (Quad gantry level AKA QGL)
  SET_DISPLAY_TEXT MSG="QGL"      # Displays info
  STATUS_LEVELING                 # Sets SB-leds to leveling-mode
  quad_gantry_level               # Levels the buildplate via QGL
  G28 Z                           # Homes Z again after QGL

  ##  Uncomment for Klicky auto-z
  CALIBRATE_Z                                 # Calibrates Z-offset with klicky
  SET_DISPLAY_TEXT MSG="Z-offset"             # Displays info

  ##  Uncomment for bed mesh (2 of 2)
  SET_DISPLAY_TEXT MSG="Bed mesh"    # Displays info
  STATUS_MESHING                     # Sets SB-leds to bed mesh-mode
  bed_mesh_calibrate                 # Starts bed mesh

  # Heats up the nozzle up to target via data from slicer
  SET_DISPLAY_TEXT MSG="Hotend: {target_extruder}c"             # Displays info
  STATUS_HEATING                                                # Sets SB-leds to heating-mode
  G1 X{x_wait} Y{y_wait} Z15 F9000                              # Goes to center of the bed
  M107                                                          # Turns off partcooling fan
  M109 S{target_extruder}                                       # Heats the nozzle to printing temp

  # Gets ready to print by doing a purge line and updating the SB-leds
  SET_DISPLAY_TEXT MSG="Printer goes brr"          # Displays info
  STATUS_PRINTING                                  # Sets SB-leds to printing-mode
  CLEAN_NOZZLE  
  VORON_PURGE
  #G0 X{x_wait - 50} Y4 F10000                      # Moves to starting point
  #G0 Z0.4                                          # Raises Z to 0.4
  #G91                                              # Incremental positioning 
  #G1 X100 E20 F1000                                # Purge line
  #G90                                              # Absolut position

[image]

jontek2, 2023-08-19:
As the logic for the V2/Trident print_start macro currently is then it heatsoaks the printer to your desired temperatur if the bed temperature is set to be above 90c. If not it will only heatsoak for 5 min (inertiaof the buildplate). This logic is there because most filament that requries a higher bed temperature also requires a heatsoak.

If you want to alter this logic so it does a full heatsoak to your desired chamber on another temperatur than above 90c, then change this logic:

`  # Checks if the bed temp is higher than 90c - if so then trigger a heatsoak.
  {% if params.BED|int > 90 %}`

Where you change "90" to the temperatur you want to have.

larkinnn, 2023-08-19:
sorry i did not explain very well. but my issue is that its still trying to do the heat soak when my bed temp is only calling 55C.

jontek2, 2023-08-19:
Oh, sorry. I've must have missunderstood oyu.

As i mentioned above it's supposed to do a mini-heatsoak for 5 mins if it's under 90c. Does it go on longer than that?

larkinnn, 2023-08-19:
ah no no. it is  me miss understanding the original docs i thought it skipped that mini 5 min heat soak as well, my bad sorry for creating a issue about this. you can delete this issue sorry about that!

jontek2, 2023-08-19:
Exactly!

No worries! Happy printing :)

closed 2023-08-19 by jontek2

airbornetrooper82573, 2023-09-08:
How do I go about disabling the 5 minute wait for items under 90 degrees?

jontek2, 2023-09-08:
You can change the following command in the macro: "G4 P300000". 300000 means milliseconds which equals to 5 minutes.  If you would like it to just hold for 30 seconds then change it to "G4 P30000".