jontek2/A-better-print_start-macro #8: Variable Heat-Soak

airbornetrooper82573, 2024-01-19:
I saw on the voron discord someone making a variable heat-soak. Basically, if you cancel a print and restart if fairly quick, it avoids having to wait for the soak for as long. 

`#Heatsoak here
  # Check if heat-soaking is needed
    {% if current_bed < target_bed - 20 %}
        M118 Bed is cold.    # display message
        SET_DISPLAY_TEXT MSG="Heating Bed: {target_bed}c"
        M190 S{target_bed}      # Set bed temperature and wait until it is reached
        M118 Heat Soak 5 minutes    # display message
        G4 P{ 5 * 60 * 1000 }       # dwell = 5  minutes
    {% elif current_bed < target_bed - 10 %}
        M118 Bed is warm.    # display message
        SET_DISPLAY_TEXT MSG="Heating Bed: {target_bed}c"
        M190 S{target_bed}      # Set bed temperature and wait until it is reached
        M118 Heat Soak 1 minute    # display message
        G4 P{ 1 * 60 * 1000 }       # dwell = 1 minute
    {% else %}
        M118 Already Hot! Heat Soak 5 seconds    # display message
        SET_DISPLAY_TEXT MSG="Heating Bed: {target_bed}c"
        M190 S{target_bed}      # Set bed temperature and wait until it is reached
        G4 P{  5 * 1000 }       # dwell = 5 seconds
    {% endif %}
  # End Heatsoak section `

I tried adding this into my print_start macro but got Error evaluating 'gcode_macro PRINT_START:gcode': jinja2.exceptions.UndefinedError: 'current_bed' is undefined. 

Would you be able to assist with incorporating something like this into the print_start? I already use your setup and just want to integrate if possible. Thank you

airbornetrooper82573, 2024-01-19:
[image]

jontek2, 2024-02-24:
This is a bit out-of-scope for this macro, but feel free to do a fork and add your own twist and take on the macro.

closed 2024-02-24 by jontek2