jontek2/A-better-print_start-macro #13: Scratched up my printer bed with deep grooves while I had no issues like this before I entered the script

Noobeley, 2024-10-31:
You should add a disclaimer to double check what configuration you have. I didn't have anything strange in my script or printer, I only followed everything from official LDO Voron 2.4 Rev D sources, never deviating with anything custom.

Entering this script into it irreversibly damaged my bed while your script never mentions any such risks because it wasn't tested with this revision.

I would put it as a heads up so no other people have this same incident happen to them.

Noobeley, 2024-11-02:
I solved the issue. It was because I started the print through the touchscreen of the printer, instead of through the slicer, which has the custom g-code you provided.

jontek2, 2024-11-02:
That shouldn’t really matter. What’s important is the file you selected to print. The G-code file with this macro implemented parses temperatures for the hotend, bed, and chamber. These values are then sent to the macro, which, based on your configuration, will execute commands to level the bed, run a bed mesh, and lay down a primeline before the print begins.

The offset ultimately depends on your saved Z-offset, which can vary depending on the probe you have on your printer. The logic may differ between probes like Omron, Klicky, Beacon, etc.

For further troubleshooting, I recommend the request_help channel on the Voron Discord. There, myself and others will be able to assist you further.

closed 2024-11-02 by jontek2