jontek2/A-better-print_start-macro #20: OrcaSlicer Modifications: Add or Replace Existing G-code?

morlan, 2025-01-31:
My current OrcaSlicer machine_start_gcode is the default:

M190 S[bed_temperature_initial_layer_single]
M109 S[nozzle_temperature_initial_layer]
PRINT_START EXTRUDER=[nozzle_temperature_initial_layer] BED=[bed_temperature_initial_layer_single]
; You can use following code instead if your PRINT_START macro support Chamber and print area bedmesh
; PRINT_START EXTRUDER=[nozzle_temperature_initial_layer] BED=[bed_temperature_initial_layer_single] Chamber=[chamber_temperature] PRINT_MIN={first_layer_print_min[0]},{first_layer_print_min[1]} PRINT_MAX={first_layer_print_max[0]},{first_layer_print_max[1]}

Do I just add your modifications or replace all of the code?  Thanks for the support!

morlan, 2025-01-31:
I read the OrcaSlicer code more carefully and noticed that the two lines at the bottom were commented out.  I'm assuming we only use one PRINT_START EXTRUDER line so I'm just replacing it all with your code.

jontek2, 2025-03-01:
This macro does not use "Demon Klipper Essentials Unified".

The status macros is for the stealthburner LED's, you can read more about them here: https://github.com/VoronDesign/Voron-Stealthburner/blob/main/Firmware/stealthburner_leds.cfg

Just follow the instructions in the README for OrcaSlicer and adapt it to your needs.

closed 2025-03-01 by jontek2