jontek2/A-better-print_start-macro #23: Klipper does not know what to do with "VALUE" command regarding nevermore pins

brownem, 2025-02-23:
# Context:
- I built my Voron 2.4r2 from an LDO rev D kit, which came with a Nevermore Micro V6. I wired everything up per their Rev D wiring guide found [here](https://docs.ldomotors.com/en/voron/voron2/wiring_guide_rev_d) in the "Connecting the Fans and the LED Strip" section.
- I am new to Vorons, Klipper, and g-code macros, but have previous experience with Prusa's and coding, so I am assuming this fault is a knowledge gap on my part, and not a legitimate error in the macros themselves (I probably named something wrong, or put something in the wrong place, etc)

# Issue:
When setting up this macro, Klipper 'rejects' portion about the nevermore having a "VALUE" function with the following error:

[image]

Here is the original portion of the printer.cfg file that I copied from LDO's github in regards to controlling the nevermore, that was tested as working prior to now:

```
[heater_fan exhaust_fan]
##  Exhaust fan - FAN3
pin: PF9
#tachometer_pin: PF8
max_power: 1.0
shutdown_speed: 0.0
kick_start_time: 5.0
heater: heater_bed
heater_temp: 60
fan_speed: 1.0
```
And here is what is after I made the recommended changes:
```
[heater_fan nevermore]
##  Exhaust fan - FAN3
pin: PF9
value: 0
shutdown_value: 0
#tachometer_pin: PF8
max_power: 1.0
shutdown_speed: 0.0
kick_start_time: 5.0
heater: heater_bed
heater_temp: 60
fan_speed: 1.0
```
Here is the `PRINT_START` macro:
```
[gcode_macro PRINT_START]
gcode:
  # This part fetches data from your slicer. Such as bed, extruder, and chamber temps and size of your printer.
  {% set target_bed = params.BED|int %}
  {% set target_extruder = params.EXTRUDER|int %}
  {% set target_chamber = params.CHAMBER|default("45")|int %}
  {% set x_wait = printer.toolhead.axis_maximum.x|float / 2 %}
  {% set y_wait = printer.toolhead.axis_maximum.y|float / 2 %}

  ##  Uncomment for Beacon Contact (1 of 4 for beacon contact)
  #SET_GCODE_OFFSET Z=0                                 # Set offset to 0

  # Home the printer, set absolute positioning and update the Stealthburner LEDs.
  STATUS_HOMING                                         # Set LEDs to homing-mode
  G28                                                   # Full home (XYZ)
  G90                                                   # Absolute position

  ##  Uncomment for bed mesh (1 of 2 for bed mesh)
  BED_MESH_CLEAR                                       # Clear old saved bed mesh (if any)

  # Check if the bed temp is higher than 90c - if so then trigger a heatsoak.
  {% if params.BED|int > 90 %}
    SET_DISPLAY_TEXT MSG="Bed: {target_bed}c"           # Display info on display
    STATUS_HEATING                                      # Set LEDs to heating-mode
    M106 S255                                           # Turn on the PT-fan

    ##  Uncomment if you have a Nevermore.
    SET_PIN PIN=nevermore VALUE=1                      # Turn on the nevermore

    G1 X{x_wait} Y{y_wait} Z15 F9000                    # Go to center of the bed
    M190 S{target_bed}                                  # Set the target temp for the bed
    SET_DISPLAY_TEXT MSG="Heatsoak: {target_chamber}c"  # Display info on display
    TEMPERATURE_WAIT SENSOR="temperature_sensor chamber" MINIMUM={target_chamber}   # Waits for chamber temp

  # If the bed temp is not over 90c, then skip the heatsoak and just heat up to set temp with a 5 min soak
  {% else %}
    SET_DISPLAY_TEXT MSG="Bed: {target_bed}c"           # Display info on display
    STATUS_HEATING                                      # Set LEDs to heating-mode
    G1 X{x_wait} Y{y_wait} Z15 F9000                    # Go to center of the bed
    M190 S{target_bed}                                  # Set the target temp for the bed
    SET_DISPLAY_TEXT MSG="Soak for 5 min"               # Display info on display
    G4 P300000                                          # Wait 5 min for the bedtemp to stabilize
  {% endif %}

  # Heat hotend to 150c. This helps with getting a correct Z-home.
  SET_DISPLAY_TEXT MSG="Hotend: 150c"                   # Display info on display
  M109 S150                                             # Heat hotend to 150c

  ##  Uncomment for Beacon contact (2 of 4 for beacon contact)
  #G28 Z METHOD=CONTACT CALIBRATE=1                     # Calibrate z offset and beacon model

  ##  Uncomment for Trident (Z_TILT_ADJUST)
  #SET_DISPLAY_TEXT MSG="Leveling"                      # Display info on display
  #STATUS_LEVELING                                      # Set LEDs to leveling-mode
  #Z_TILT_ADJUST                                        # Level the printer via Z_TILT_ADJUST
  #G28 Z                                                # Home Z again after Z_TILT_ADJUST

  ##  Uncomment for V2.4 (Quad gantry level AKA QGL)
  SET_DISPLAY_TEXT MSG="Leveling"                      # Display info on display
  STATUS_LEVELING                                      # Set LEDs to leveling-mode
  QUAD_GANTRY_LEVEL                                    # Level the printer via QGL
  G28 Z                                                # Home Z again after QGL

  ##  Uncomment for bed mesh (2 of 2 for bed mesh)
  SET_DISPLAY_TEXT MSG="Bed mesh"                      # Display info on display
  STATUS_MESHING                                       # Set LEDs to bed mesh-mode
  BED_MESH_CALIBRATE                                   # Start the bed mesh (add ADAPTIVE=1) for adaptive bed mesh

  ## Uncomment for Beacon Contact (3 of 4 for beacon contact)
  #G28 Z METHOD=CONTACT CALIBRATE=0                     # Calibrate z offset only with hot nozzle

  # Heat up the hotend up to target via data from slicer
  SET_DISPLAY_TEXT MSG="Hotend: {target_extruder}c"     # Display info on display
  STATUS_HEATING                                        # Set LEDs to heating-mode
  G1 X{x_wait} Y{y_wait} Z15 F9000                      # Go to center of the bed
  M107                                                  # Turn off partcooling fan
  M109 S{target_extruder}                               # Heat the hotend to set temp

  ##   Uncomment for Beacon Contact (4 of 4 for beacon contact)
  #SET_GCODE_OFFSET Z=0.06                              # Add a little offset for hotend thermal expansion

  # Get ready to print by doing a primeline and updating the LEDs
  SET_DISPLAY_TEXT MSG="Printer goes brr"               # Display info on display
  STATUS_PRINTING                                       # Set LEDs to printing-mode
  G0 X{x_wait - 50} Y4 F10000                           # Go to starting point
  G0 Z0.4                                               # Raise Z to 0.4
  G91                                                   # Incremental positioning 
  G1 X100 E20 F1000                                     # Primeline
  G90                                                   # Absolute position
```
And the `PRINT_END` macro:
```
[gcode_macro PRINT_END]
#   Use PRINT_END for the slicer ending script - please customise for your slicer of choice
gcode:
    # safe anti-stringing move coords
    {% set th = printer.toolhead %}
    {% set x_safe = th.position.x + 20 * (1 if th.axis_maximum.x - th.position.x > 20 else -1) %}
    {% set y_safe = th.position.y + 20 * (1 if th.axis_maximum.y - th.position.y > 20 else -1) %}
    {% set z_safe = [th.position.z + 2, th.axis_maximum.z]|min %}
    
    SAVE_GCODE_STATE NAME=STATE_PRINT_END
    
    M400                           ; wait for buffer to clear
    G92 E0                         ; zero the extruder
    G1 E-5.0 F1800                 ; retract filament
    
    TURN_OFF_HEATERS
    
    G90                                      ; absolute positioning
    G0 X{x_safe} Y{y_safe} Z{z_safe} F20000  ; move nozzle to remove stringing
    G0 X{th.axis_maximum.x//2} Y{th.axis_maximum.y - 2} F3600  ; park nozzle at rear
    M107                                     ; turn off fan

    SET_PIN PIN=nevermore VALUE=0
    
    BED_MESH_CLEAR
    RESTORE_GCODE_STATE NAME=STATE_PRINT_END
```
If it matters at all, I plan to eventually add additional bed fans, and supplement my Nevermore Micro V6 with a Nevermore StealthMax at some point in the future, so I will probably be adding in an 'exhaust_fan' section that gets called in the `PRINT_START` and `PRINT_END` macros as well. Just not now.

Any help you can provide would be greatly appreciated! Thank you!

brownem, 2025-02-24:
Alright, and update: I made some progress, but there is still some odd bugs.

Klipper starts up just fine and prints start according to this macro, but it reports a fair number of warnings. I changed the code for the Nevermore from this:
```
[heater_fan nevermore]
##  Exhaust fan - FAN3
pin: PF9
value: 0
shutdown_value: 0
#tachometer_pin: PF8
max_power: 1.0
shutdown_speed: 0.0
kick_start_time: 5.0
heater: heater_bed
heater_temp: 60
fan_speed: 1.0
```
To this:
```
[heater_fan nevermore]
##  Exhaust fan - FAN3
pin: PF9
#tachometer_pin: PF8
max_power: 1.0
shutdown_speed: 0.0
kick_start_time: 5.0
heater: heater_bed
heater_temp: 60
fan_speed: 1.0
```

Essentially changing is back to the 'LDO-standard' code, except I changed only `exhaust_fan` to `nevermore`. That's it. By doing this, Klipper reboots and I can sent test prints to it just fine. IT even starts and stops the nevermore with printing (but I haven't quite figured out how it is doing this just yet). Except it reports the following faults during printing:

- `Unknown command:"STATUS_HOMING"`
- `Unknown command:"STATUS_HEATING"`
- `Unknown command:"STATUS_LEVELING"`
- `Unknown command:"STATUS_MESHING"`
- `Unknown command:"STATUS_HEATING"`
- `Unknown command:"STATUS_PRINTING"`
- `The value 'nevermore' is not valid for PIN` (2x in rapid succession according to console outputs)

And interestingly, it reported the print as "canceled" instead of completed. So it looks like now it doesn't really know what to do with any of the "status" flags in the macro, and it still doesn't know how to handle the Nevermore control (possibly because it is being controlled by the standard LDO code some other way, and this is breaking this macro?)

I'm going to keep poking at it.

jontek2, 2025-02-25:
Hi,

The first issue you posted is related to you not having the nevermore set up as described in the README. You had it configured as a heater_fan and not as a output_pin. 

The Unknown command:"STATUS_HOMING" is due to not having the StealthBurner LED logic inplace, you can get them here https://github.com/VoronDesign/Voron-Stealthburner/blob/main/Firmware/stealthburner_leds.cfg

brownem, 2025-02-27:
That solved the status messages, but I am still running into issues with the Nevermore.

# Just to cover what I did for the status messages for anyone else who comes looking in the future:

Then for the Stealthburner Neopixel configuration, I just loaded in that file, updated the pin (and double checked the specific RGB configuration against my printer.cfg data) and called it up in printer.cfg file exactly as described in that link. The only changes I needed to make were:

1. Comment out the original Neopixel code in the LED control section of the original LDO code (also where the pin, chain count, and color order info can be found)
2. The original gcode macros from LDO: `_LOGO_PENDING`, `_LOGO_READY`, `_LOGO_OFF`, `_HEADLIGHT_ON`, `_HEADLIGHT_OFF`
3. Basically anywhere else that called out the `rgb_light` value in the printer.cfg file

# Nevermore issue

I did not understand that `output_pin` was a Klipper-specific command, and not just a placeholder pseudo-code. Now that I understand that, my nevermore configuration looks like:

```
[output_pin nevermore]
## Exhaust fan - FAN3
pin: PF9
value: 0
shutdown_value: 0
```
As a bonus, I understand more about `heater_fan` in Klipper configurations, which will come in handy when I add more under-bed fans and a Nevermore Stealth Max.

However, while this clears the errors in the Klipper console output and firmware startups, the fans themselves do not actually startup with a print with this code. I think they might be stalled out on start up. Looking through the Klipper docs for [`heater_fan`](https://www.klipper3d.org/Config_Reference.html#heater_fan), you can see options for thinks like `kick_start_time`, but [`output_pin`](https://www.klipper3d.org/Config_Reference.html#output_pin) does not seem to have the same options to help get fans started past their stall speed.

Unless I am missing something else in this start macro that only turns the nevermore on when the chamber temp is above a certain set point?

brownem, 2025-02-27:
Interestingly, by putting the nevermore on an `output_pin` it now shows up under the Klipper Screen option: ⚙️More > # Pins > nevermore, with a binary switch to turn them on and off. And that switch does turn it on/off, without a stall out. So the fans can be physically turned on using `output_pin`.

Guess I am hunting another bug in my implementation of my start macro.

brownem, 2025-03-01:
So... I came up with a solution to get this macro to work and the Nevermore to turn on/off with bed usage (including PID tunes and other heating cycles not triggered through this macro), and that was to comment out the section about Nevermore in this macro, and go back to using `heater_fan` to control it. 

So my Nevermore section looks like this:
```
[heater_fan nevermore]
##  Exhaust fan - FAN3
pin: PF9
tachometer_pin: PF8
max_power: 1.0
shutdown_speed: 0.0
kick_start_time: 5.0
heater: heater_bed
heater_temp: 60
fan_speed: 1.0
```

My `PRINT_START` macro looks like this:
```
[gcode_macro PRINT_START]
gcode:
  # This part fetches data from your slicer. Such as bed, extruder, and chamber temps and size of your printer.
  {% set target_bed = params.BED|int %}
  {% set target_extruder = params.EXTRUDER|int %}
  {% set target_chamber = params.CHAMBER|default("45")|int %}
  {% set x_wait = printer.toolhead.axis_maximum.x|float / 2 %}
  {% set y_wait = printer.toolhead.axis_maximum.y|float / 2 %}

  ##  Uncomment for Beacon Contact (1 of 4 for beacon contact)
  #SET_GCODE_OFFSET Z=0                                 # Set offset to 0

  # Home the printer, set absolute positioning and update the Stealthburner LEDs.
  STATUS_HOMING                                         # Set LEDs to homing-mode
  G28                                                   # Full home (XYZ)
  G90                                                   # Absolute position

  ##  Uncomment for bed mesh (1 of 2 for bed mesh)
  BED_MESH_CLEAR                                       # Clear old saved bed mesh (if any)

  # Check if the bed temp is higher than 90c - if so then trigger a heatsoak.
  {% if params.BED|int > 90 %}
    SET_DISPLAY_TEXT MSG="Bed: {target_bed}c"           # Display info on display
    STATUS_HEATING                                      # Set LEDs to heating-mode
    M106 S255                                           # Turn on the PT-fan

    ##  Uncomment if you have a Nevermore.
    #SET_PIN PIN=nevermore VALUE=1                      # Turn on the nevermore

    G1 X{x_wait} Y{y_wait} Z15 F9000                    # Go to center of the bed
    M190 S{target_bed}                                  # Set the target temp for the bed
    SET_DISPLAY_TEXT MSG="Heatsoak: {target_chamber}c"  # Display info on display
    TEMPERATURE_WAIT SENSOR="temperature_sensor chamber" MINIMUM={target_chamber}   # Waits for chamber temp

  # If the bed temp is not over 90c, then skip the heatsoak and just heat up to set temp with a 5 min soak
  {% else %}
    SET_DISPLAY_TEXT MSG="Bed: {target_bed}c"           # Display info on display
    STATUS_HEATING                                      # Set LEDs to heating-mode
    G1 X{x_wait} Y{y_wait} Z15 F9000                    # Go to center of the bed
    M190 S{target_bed}                                  # Set the target temp for the bed
    SET_DISPLAY_TEXT MSG="Soak for 5 min"               # Display info on display
    G4 P300000                                          # Wait 5 min for the bedtemp to stabilize
  {% endif %}

  # Heat hotend to 150c. This helps with getting a correct Z-home.
  SET_DISPLAY_TEXT MSG="Hotend: 150c"                   # Display info on display
  M109 S150                                             # Heat hotend to 150c

  ##  Uncomment for Beacon contact (2 of 4 for beacon contact)
  #G28 Z METHOD=CONTACT CALIBRATE=1                     # Calibrate z offset and beacon model

  ##  Uncomment for Trident (Z_TILT_ADJUST)
[... 49 lines cut ...]
    
    SAVE_GCODE_STATE NAME=STATE_PRINT_END
    
    M400                           ; wait for buffer to clear
    G92 E0                         ; zero the extruder
    G1 E-5.0 F1800                 ; retract filament
    
    TURN_OFF_HEATERS
    
    G90                                      ; absolute positioning
    G0 X{x_safe} Y{y_safe} Z{z_safe} F20000  ; move nozzle to remove stringing
    G0 X{th.axis_maximum.x//2} Y{th.axis_maximum.y - 2} F3600  ; park nozzle at rear
    M107                                     ; turn off fan

    #SET_PIN PIN=nevermore VALUE=0
    
    BED_MESH_CLEAR
    RESTORE_GCODE_STATE NAME=STATE_PRINT_END
```

This:
1. Turns my Nevermore Micro V6 on/off with prints and any other bed heating activities, regardless of macros, via the functions built into `heater_fan` and associating it with the heater bed itself
2. Still runs homing, heat soaking, QGL, bed meshing, neopixel, and statuses

Which seems to cover all the bases for printing with under-bed cooling/heat circulation in the chamber.

closed 2025-03-01 by brownem